LabPy/lantz #40: VisaDriver fails without a library_path keyword

gdavis-xilinx, 2015-01-30:
In package lance.visa, line 37
        library_path = kwargs.get('library_path', None)
should be:
        library_path = kwargs.get('library_path', '')
to conform with the default of the visa.ResourceManager.init().

Or, to be less brittle:
        library_path = kwargs.get('library_path', None)
        if library_path:
                  manager = visa.ResourceManager(library_path)
        else:
                  manager = visa.ResourceManager()

hgrecco, 2015-01-31:
Thanks for the bug report. This is due to a change in PyVISA. Care to submit a PR?

Also notice that the VISA driver is going to go away in the future. The MesseageBasedDriver (currently in develop) and to debut in 0.3 will takes its place.

gdavis-xilinx, 2015-02-02:
PR? Unknown acronym.

hgrecco, 2015-02-02:
PR = Pull Request. You fork de repo, introduce your changes, commit, push and then make a pull request so I can merge and is properly attributed to to you.

gdavis-xilinx, 2015-02-02:
Done.

closed 2015-02-05 by hgrecco