LabPy/lantz #43: Splitting Lantz

hgrecco, 2015-02-08:
The proposal is to leave the current Lantz repo as it is by now and create 3 new repos in LabPy

(Please bikeshed the names of the repos. These names are important as they will become the names of independent packages on PyPI)

**lantz_core**:
- action.py
- errors.py
- feat.py
- foreign.py
- log.py
- messagebased.py
- processors.py
- stats.py

(+ corresponding tests and docs)

**lantz_drivers**
- drivers/*

(But removing the legacy folder)
(+ corresponding tests and docs)

**lantz_qt**
- ui/*
- util/*

(+ corresponding tests and docs)

After the core is ready, a new package (`lantz_enaml`) will be created.

In the future and to help the user, a Lantz package that installs the core, drivers and qt/enaml will be created with a entry points to the docs.

MatthieuDartiailh, 2015-02-08:
If we integrate the Eapii architecture will also have in the core : 
- has_feature.py
- subsystem.py
- channel.py
- base_driver.py (or base_instrument.py)
- I would like to also keep the i_properties folder
- I would also suggest moving the backends (foreign, message_based (is this always VISA), etc into their own packages.

The simulators seems a bit obsolete to me if we get a working pyvisa-sim, save for some foreign but I am not sure we will ever find time to write them anyway.
I am fine with the other names.

hgrecco, 2015-02-08:
Right now the simulators can do a few things that `pyvisa-sim` cannot do (We can teach it!) In any case we can look at that later.

Do you mean a backends folder with foreign.py and messagebased.py?

MatthieuDartiailh, 2015-02-08:
Basically yes but we might need to add a registry based also and perhaps other later on.

hgrecco, 2015-02-08:
One of Lantz users also has COM object based driver that he plans to contribute.

MatthieuDartiailh, 2015-02-08:
I think it might also be useful to add a unit module dealing with how Lantz interacts with Pint (basically allowing to set the UnitRegistry used), you can have a look at Eapii unit module. And in the same idea a way to interact with pyvisa which can be in backends.
What do we decide about the processors ? Do we go the Eapii route with the IProperty (which becomes Feature) ?

hgrecco, 2015-02-09:
+1 about making the unit module explicitely

I think we get rid of processors in the current form but keep the functionality in the way eapii provides them. There a few things which I am not so sure if IProperty is (currently) capable. Basically:
- Changing units of a Property: this is useful for instruments in which you can switch between units and this results in different ranges.
- Using a variable of the object (only known as instantiation time) for validation.

hgrecco, 2015-02-09:
I have trimmed Lantz and splitted in three packages. I deleted some files in each repo (those we are sure that we do not need) and I left others (some that we need, some we will delete as soon as we have a replacement). Let's address any specific issue in the corresponding package.

closed 2015-02-09 by hgrecco

MatthieuDartiailh, 2015-02-09:
I will try to keep your remark in my mind. Changing the unit would mean the unit is stored somewhere else we could use a '@unit_attribute' syntax when declaring a unit to let know the Feature where to look for. 
About your second point it is already possible to do it for ranges by specifying the name of the range to use, once again perhaps it is worth extending. I will try to keep that in mind.

hgrecco, 2015-02-09:
Great. By the way, I opened a wiki section for organizing the way to push into the LabPy repo.

MatthieuDartiailh, 2015-02-09:
I noticed and updated it to include the PEP8 standard (we will perhaps release some constraints later on but I think it is nice to mention it).

vascotenner, 2015-02-16:
I'm currently developing some drivers in a fork of the old lantz project. How can I move this branches to forks of the new lantz_driver repro?

hgrecco, 2015-02-16:
@vascotenner Good to se eyou here. The internal structure of drivers is going to be enriched by a few more features. You can either add a PR to LabPy/Lantz and then we can migrate all the drivers all together 
You are off course most welcome to join the discussions.