LabPy/lantz #45: Exceptions in the Core

hgrecco, 2015-02-08:
I think is a really good opportunity to think again the exception hierarchy in Lantz.

Right now we have the following errors:
- InvalidCommand: The command send to the instrument was invalid.
- LantzTimeoutError: A timeout occurred. Used to have a single error abstracting the differences in lower level libraries
- InstrumentError: Generic error
- NotSupportedError: The interface is not supported by this instrument

Maybe it could be reworked as
LantzError: Generic Error

And deriving from this:
- InvalidCommand
- TimeoutError
- InterfaceNotSupported

Another question? Do we need to reraise pyvisa (or other libraries) exceptions?

MatthieuDartiailh, 2015-02-08:
I like your proposal. I will have to think about what to do about the notion of communication safety I use in Eapii (and which a requirement of some of my users). The idea is basically that is an operation fails without apparent reason (unexpected timeout) it may be worth trying again after reopening the connection because some instruments are dumb.

hgrecco, 2015-02-08:
Some instruments are indeed dumb. But we need to make this somehow opt-in. Maybe with a retry flag?

MatthieuDartiailh, 2015-02-08:
Agreed for the time being, retry behavior is specified at the IProperty level and is then specified for all driver of the type, we could have a boolean flag at the instance level to de-activate this behavior.