LabPy/lantz #48: Unable to use lantz.ui.app.Frontend

operte, 2015-02-23:
I'm new to Lantz but I found its features to match the requirements of an experiment I'm currently working on. I'm basing most of my development on the tutorial. I've already written the drivers for my device and wanted to start building a GUI. For that, I started the rich-app tutorial. However, I immediately ran into an error after running the first piece of code:
```javascript
# We import a helper function to start the app
from lantz.ui.app import start_gui_app

# The block consists of two parts the backend and the frontend
from lantz.ui.blocks import FeatScan, FeatScanUi

# An this you know already
from lantz.drivers.examples import LantzSignalGenerator

with LantzSignalGenerator('TCPIP::localhost::5678::SOCKET') as fungen:

    # Here we instantiate the backend setting 'frequency' as the Feat to scan
    # and specifying in which instrument
    app = FeatScan('frequency', instrument=fungen)

    # Now we use the helper to start the app.
    # It takes a Backend instance and a FrontEnd class
    start_gui_app(app, FeatScanUi)
```
```javascript
Traceback (most recent call last):

  File "<ipython-input-14-d6e2c2282680>", line 1, in <module>
    runfile('C:/[...]/test.py', wdir='C:/Users/simon/Documents/JM/Santec TSL-510/PyControl')

  File "C:\[...]\Anaconda\envs\py34\lib\site-packages\spyderlib\widgets\externalshell\sitecustomize.py", line 601, in runfile
    execfile(filename, namespace)

  File "C:\[...]\Anaconda\envs\py34\lib\site-packages\spyderlib\widgets\externalshell\sitecustomize.py", line 80, in execfile
    exec(compile(open(filename, 'rb').read(), filename, 'exec'), namespace)

  File "C:/[...]/Documents/JM/Santec TSL-510/PyControl/test.py", line 20, in <module>
    start_gui_app(app, FeatScanUi)

  File "C:\[...]\Anaconda\envs\py34\lib\site-packages\lantz\ui\app.py", line 242, in start_gui_app
    frontend = frontend_class(backend=backend)

  File "C:\[...]\Anaconda\envs\py34\lib\site-packages\lantz\ui\app.py", line 131, in __init__
    filename = os.path.dirname(inspect.getfile(cls))

  File "C:\[...]\Anaconda\envs\py34\lib\inspect.py", line 524, in getfile
    raise TypeError('{!r} is a built-in class'.format(object))

TypeError: <module 'builtins' (built-in)> is a built-in class
```

It seems the problem resides on a section of `lantz.ui.app.Frontend`:
```javascript
class Frontend(QtGui.QMainWindow, metaclass=_FrontendType):

    frontends = {}

     # a declarative way to indicate the user interface file to use.
    gui = None

    # connect widgets to instruments using connect_setup
    auto_connect = True

    def __init__(self, parent=None, backend=None):
        super().__init__(parent)

        self._backend = None

        if self.gui:
            for cls in self.__class__.__mro__:
                filename = os.path.dirname(inspect.getfile(cls))       # <<<<<< This line!
                filename = os.path.join(filename, self.gui)
                if os.path.exists(filename):
                    logger.debug('{}: loading gui file {}'.format(self, filename))
                    self.widget = QtGui.loadUi(filename)
                    self.setCentralWidget(self.widget)
                    break
            else:
                raise ValueError('{}: loading gui file {}'.format(self, self.gui))
    
    # ...
```
The elements inside `Frontend.__class__.__mro__` are:
```javascript
(lantz.ui.app._FrontendType,
 PyQt4.QtCore.pyqtWrapperType,
 sip.wrappertype,
 type,
 object)
```
From this it seems the problem is when `for cls in self.__class__.__mro__` goes through `type` and `object`. I think that since these two classes are built-ins, `inspect.getfile(cls)` raises the `TypeError` exception.
I'm not certain if this problem is specific to my system or if it is a general issue. Could it be an incompatibility between lantz and my current libraries? I'm using Python 3.4 and I think that I have every library up-to-date.

operte, 2015-02-23:
My naive approach was to simply remove the last two elements of `Frontend.__class__.__mro__`, hoping to avoid `inspect.getfile(cls)` to operate on `type` and `object`. I thought about doing it like this:
```javascript
class Frontend(QtGui.QMainWindow, metaclass=_FrontendType):
    # ...
    def __init__(self, parent=None, backend=None):
        # ...
        if self.gui:
            for cls in self.__class__.__mro__[0:-2]: # <<< Change here.
                # ...
```
This (certainly) naive modification eliminates the previous error message. However it does raise a new problem:
```javascript
Traceback (most recent call last):

  File "<ipython-input-1-d6e2c2282680>", line 1, in <module>
    runfile('C:/[...]/test.py', wdir='C:/[...]')

  File "C:\[...]\Anaconda\envs\py34\lib\site-packages\spyderlib\widgets\externalshell\sitecustomize.py", line 601, in runfile
    execfile(filename, namespace)

  File "C:\[...]\Anaconda\envs\py34\lib\site-packages\spyderlib\widgets\externalshell\sitecustomize.py", line 80, in execfile
    exec(compile(open(filename, 'rb').read(), filename, 'exec'), namespace)

  File "C:/[...]/test.py", line 20, in <module>
    start_gui_app(app, FeatScanUi)

  File "C:\[...]\Anaconda\envs\py34\lib\site-packages\lantz\ui\app.py", line 242, in start_gui_app
    frontend = frontend_class(backend=backend)

  File "C:\[...]\Anaconda\envs\py34\lib\site-packages\lantz\ui\app.py", line 139, in __init__
    raise ValueError('{}: loading gui file {}'.format(self, self.gui))

ValueError: FeatScanUi: loading gui file scan.ui
```
It seems to me that now, the condition `if os.path.exists(filename)` in `Frontend.__init__` is never `true` for the elements in `self.__class__.__mro__[0:-2]`. However, I have manually tested every element in `Frontend.__class__.__mro__[0:-2]`, namely `(lantz.ui.app._FrontendType, PyQt4.QtCore.pyqtWrapperType, sip.wrappertype)` and the output of applying `os.path.exists(filename)` on each one is `True`. That said, I think that either my naive approach didn't work or I have reached yet another issue. Any tips to solving this are very much welcome!

hgrecco, 2015-02-23:
@operte Thanks for the bug report. I have not seen this problem before. Can you tell me which Python version and PyQt version are you using?

operte, 2015-02-24:
@hgrecco I am using Python 3.4.2, PyQt4 4.10.4 and Qt 4.8.6. I don't have PySide installed so I'm not certain that the issue you referenced is connected to this problem.

hgrecco, 2015-02-25:
I have roughly the same setup. and it works for me. I will try to mimick your installation, try it and let you know.

operte, 2015-02-25:
@hgrecco Also I am running on Windows. Could it be related?

hgrecco, 2015-02-25:
I do not use Windows but I have tried it on windows. I am wondering if this is because you are running within IPython. Can you try it outside?

operte, 2015-02-25:
Running this on a Python shell results in the same error message. I am going to try to install a new conda environment with the minimum amount of libraries necessary to check if my problem persists.

hgrecco, 2015-02-25:
I have access to a Windows computer tomorrow. I will try and let you know.

jflima11br, 2015-05-20:
Hi! I found the same problem that @operte. I'm using python 3.4.3 on windows. Any tips about how to solve that?

hgrecco, 2015-05-20:
@operte and @jflima11br are you both usiing PyQt or PySIDE?

jflima11br, 2015-05-21:
I'm using PyQt.

bizarri, 2015-07-23:
I am not sure if you are still looking for an answer but I had a similar issue than the ones reported by @operte and @jflima11br. 
My problem was the missing ui files for the different blocks (scan.ui, loop.ui...). I just added the files taken from lantz_qt/ui/blocks in the lantz/ui/blocks folder and now everything is working.
Hope it helps.

jflima11br, 2015-07-23:
Hi @bizarri! Same problem here... everything is working. Thank you!

mpwer, 2017-04-12:
Thanks @bizarri !
Solved it for me