LabPy/lantz #51: Error with logging, Style not defined

joeadam, 2015-04-09:
Using lantz 0.3 on Windows (Python 3.4.3 :: Anaconda 2.2.0 (32-bit).

When I try to use logging, 

    from lantz.log import log_to_screen, DEBUG
    log_to_screen(DEBUG)

I get an error on every log print

--- Logging error ---
Traceback (most recent call last):
  File "C:\Anaconda3\lib\logging\__init__.py", line 978, in emit
    msg = self.format(record)
  File "C:\Anaconda3\lib\logging\__init__.py", line 828, in format
    return fmt.format(record)
  File "C:\Anaconda3\lib\site-packages\lantz\log.py", line 148, in color_format
    parts[0] = bef + self.colorize(dur, record) + aft
  File "C:\Anaconda3\lib\site-packages\lantz\log.py", line 136, in colorize
    return color + message + Style.RESET_ALL
NameError: name 'Style' is not defined
Call stack:
  File "k2308.py", line 88, in <module>
    print(inst.output_enable[1])
  File "C:\Anaconda3\lib\site-packages\lantz\driver.py", line 355, in __exit__
    self.finalize()
  File "C:\Anaconda3\lib\site-packages\lantz\messagebased.py", line 332, in fina
lize
    super().finalize()
  File "C:\Anaconda3\lib\site-packages\lantz\action.py", line 129, in call
    instance.log_info('{} returned {}', name, out)
  File "C:\Anaconda3\lib\site-packages\lantz\driver.py", line 304, in log_info
    self.log(logging.INFO, msg, *args, **kwargs)
  File "C:\Anaconda3\lib\site-packages\lantz\driver.py", line 296, in log
    logger.log(level, msg, *args, extra=self.log_extra)
Message: '{} returned {}'
Arguments: ('finalize', None)

If I import colorama manually, it imports fine, and Style is present.

hgrecco, 2015-04-13:
Thanks for the report. We are anyway reworking but I will try to fix it before if I have some time.

ntvandehey, 2015-04-15:
Joeadam-
The same problem was driving me nuts for hours, trying to figure out if had installed something wrong. I see you say "If I import colorama manually, it imports fine, and Style is present."
What exactly do you do to import colorama manually? Are you running through the tutorial?
Thanks.

ntvandehey, 2015-04-16:
figured it out... 

insert this in log.py, just after "import threading" (line 19 for me)

    import colorama
    from colorama import Fore, Back, Style

joeadam, 2015-04-24:
Fixes it for me as well

Joseph Adam, Technology Developer

Sentec Ltd              phone:  +44(0) 1223 303800
5 The Westbrook Centre  fax:    +44(0) 1223 303801
Milton Road              
Cambridge               email:  jadam@sentec.co.uk
CB4 1YG, UK             web:    www.sentec.co.uk

This email is confidential. If you have received it in error, please notify Sentec Ltd UK at postmaster@sentec.co.uk immediately,
delete it from your system and note that you may not copy, distribute or use its contents.

Sentec Limited is registered at the above address UK Company Number 3452194.

z-acl, 2016-08-23:
Hello,
I have got the same problem and your solution fixed it.