LabPy/lantz #52: how to set path to visa driver

phsdv, 2015-06-10:
I am trying out Lantz, but it fails with a timeout when accessing my instrument.
When using pyvisa it works but I have to set the path to the visa driver in the resource manager. How do I tell Lantz which visa driver to use?

Or in other words: No documentation exists (or I cannot find it) to specify in Lantz the path to the driver that pyvisa should use.

phsdv, 2015-06-10:
Oops, found out that timeout was caused by wrong command. replacing '?IDN' with '*IDN?' solved the timeout.

However still do not know how to specify a specific visa driver, for example the NI-Visa one or the Agilent visa driver.

hgrecco, 2015-06-10:
The easiest way is just to define it in `.pyvisarc` as discussed [here](http://pyvisa.readthedocs.org/en/master/configuring.html) Lantz will honor that. We will add an option to do it in Lantz later.