LabPy/lantz #53: Create a driver with a different resource manager

bizarri, 2015-07-13:
I am trying to create a message based drivers using the pyvisa-py backend. 
I am obviously missing something, somewhere. 
How can I define the backend resource manager that will be used in lantz?

I tried to create a class for my driver something looking like:
class LantzSignalGenerator(MessageBasedDriver):
    """Lantz Signal Generator.
    """

    DEFAULTS = {'ASRL': {'read_termination': '\n', 'baud_rate': 9600}}
    __resource_manager = visa.ResourceManager("@py")
.....

and I have the error:
Traceback (most recent call last):
  File "mydriver.py", line 20, in <module>
    inst = LantzSignalGenerator('ASRL/dev/ttyACM0::INSTR')
  File "/home/bizarri/bin/miniconda3/lib/python3.4/site-packages/lantz/messagebased.py", line 299, in __init__
    self.__resource_manager = get_resource_manager()
  File "/home/bizarri/bin/miniconda3/lib/python3.4/site-packages/lantz/messagebased.py", line 40, in get_resource_manager
    _resource_manager = visa.ResourceManager()
  File "/home/bizarri/bin/miniconda3/lib/python3.4/site-packages/pyvisa/highlevel.py", line 1488, in __new__
    visa_library = open_visa_library(visa_library)
  File "/home/bizarri/bin/miniconda3/lib/python3.4/site-packages/pyvisa/highlevel.py", line 1460, in open_visa_library
    return cls(argument)
  File "/home/bizarri/bin/miniconda3/lib/python3.4/site-packages/pyvisa/highlevel.py", line 96, in __new__
    raise OSError('Could not open VISA library:\n' + '\n'.join(errs))
OSError: Could not open VISA library:

That I interpret as the creation trying to find the "standard" ni backend (not installed).

When I directly changing the backend in Lantz messagebased.py file:
#_resource_manager = None
_resource_manager = visa.ResourceManager("@py")

Everything works fine...

In the __init__ of Lantz messagebased.py, it seems that the first thing done to look for the resource_manager via get_resource_manager()
        self.__resource_manager = get_resource_manager()
Does it make sense??? Define like this, _resource_manager will always be None?
   global _resource_manager
    if _resource_manager is None:
        _resource_manager = visa.ResourceManager()
    return _resource_manager

I am certainly missing something trivial but unfortunately it is beyond my limited understanding of python. 

Thanks in advance for your time and help.

vascotenner, 2015-11-09:
I have the same issue. I would even like to have different resource managers for different drivers

vascotenner, 2017-02-23:
I think this can be bypassed by setting the resourcemanager:

```
import lantz.messagebased as mb
mb._resource_manager = visa.ResourceManager("@py")
```

And then start the device.