LabPy/lantz #73: Python2

sur83, 2016-09-19:
I strongly need to use python2 instead of python3. Is there any possibilities to convert this code into python2 code?

sur83, 2016-09-19:
I am making a project which control a variety of cameras with a common python program (when finished, I want to share it). I have developed it in python 2 from the beginning, and now it is in a very advanced state. That is the reason to the needed of use python2.
I saw that with this program I would be able to control a xenic camera, so in this case the work is already done.

vascotenner, 2016-09-19:
Note that Lantz currently supports xenics cameras via the legacy Xcamera driver. This driver is not available for windows 7 and higher. This driver supports some older (line-camera) models, that are not supported anymore in the new driver. The differences between the new and old driver are not large, and the current driver is a good starting point to implement the new driver. If you implement the new driver, I am very happy to use it!

sur83, 2016-09-19:
Thank you very much for your comments. 3to2 is not working. I will try to adapt the code manually, if success I will share it.